whaleygeek/bitio #1: sense.py doesn't show accelerometer during execution but when disconnecting micro:bit

titimoby, 2017-06-09:
The sense.py is working for the buttons, A and B are displayed on the micro:bit.
No values are printed on the console from where I launched the script.
If I disconnect the micro:bit, all the print statements appears (and then the SerialException which is right)

I've done this with python 3.6.1 on Windows in Git Shell

whaleygeek, 2017-06-09:
oh, that's odd.

I just tried this in python 2 and python 3 and I get this output:

[image]

I forgot to add a step to the install instructions for windows - do you have the mbed serial driver installed?

Instructions here: https://support.microbit.org/support/solutions/articles/19000022103-how-do-i-use-the-serial-port-with-a-micro-bit-on-windows

whaleygeek, 2017-06-09:
I've yet to test this library thoroughly on all platforms and haven't had a chance to test on windows yet. I will try next week.

What version of windows are you using?

Also, can you check if you have a pyserial in your distpackages?

e.g. if you just run
```python3```
from the shell and then
```import serial```
then
```print(serial.VERSION)
```

what do you get?

whaleygeek, 2017-06-09:
I have packaged a version of pyserial 3.3 in the microbit folder and the hope is that it imports that, but I may have to add some extra logic to force it to import that if you already have a pyserial in your distpackages - there are various incompatible changes between each version of pyserial, and I'm trying to force this to use a specific version (i.e. a version I have already tested against).

titimoby, 2017-06-10:
So first, it's not a major issue as I used Windows just because I was just coming from a game with my son.
But still, as I have this Windows box, I can do tests before booting back to my beloved Ubuntu ;)
In order:
1) I have mbed driver, I already used it for Mu editor
2) output => 3.3
If you want me to do other operations, just ask, it will be my contribution to this lovely project.

titimoby, 2017-06-10:
As I had a very small test programm to exchange between micro:bit and PC I tried it.
And it failed the same way : nothing and as soon as I interrupt, all print statement result appears.

As I mentionned, I worked in Git shell which is not the normal command line.
So I launched my script then yours and everything worked as expected.

The conclusion is we need to use the official command line for serial interactions.
That being said, I boot back to Ubuntu ;)

whaleygeek, 2017-06-14:
Ok, thanks for the update! At least if anyone else uses this in gitshell this issue will help them understand what the problem is. Thanks for trying it!

closed 2017-06-14 by whaleygeek