whaleygeek/bitio #3: Intermittent error on Raspberry Pi

martinohanlon, 2017-06-20:
I did a little bit of testing on a Pi.

I received this error a couple of times (I wasnt able to recreate it consistently though, so not sure how much to worry about it):

```
Traceback (most recent call last):
  File "button.py", line 10, in <module>
    if microbit.button_a.was_pressed():
  File "/home/pi/bitio/src/microbit/api.py", line 42, in was_pressed
    r = self.parent.cmd("print(%s.was_pressed())" % self.name)
  File "/home/pi/bitio/src/microbit/api.py", line 24, in cmd
    r = self.repl.wait_response()
  File "/home/pi/bitio/src/microbit/repl/repl.py", line 167, in wait_response
    self.receive("OK", timeout=1, min_length=2)
  File "/home/pi/bitio/src/microbit/repl/repl.py", line 64, in receive
    ch = self.readch()
  File "/home/pi/bitio/src/microbit/repl/repl.py", line 25, in rx
    data = ser.read(1)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 460, in read
    raise SerialException('device reports readiness to read but returned no data (device disconnected?)')
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected?)
```

whaleygeek, 2017-06-30:
Sounds like a bug in pyserial. Will keep it on the list.

whaleygeek, 2017-07-06:
Informed by this....

https://github.com/pyserial/pyserial/issues/86

I suspect you have an old python process running in another window, or more likely in an orphaned shell that is running in the background. The Raspberry Pi is notorious for leaving python programs running in the background if you accidentally click the X in the window at the wrong time. We often see this with our LED flashing program in our alarm system resource with the IET kits where kids click the X and then their LED is flashing in seemingly random patterns.

The way to diagnose this is to do this from another terminal prompt and see what other python processes are still running:

```ps -ax | grep python```

So, unfortunately, I think this might be classified as "user error"

@martinohanlon perhaps next time you see this you can run the above diagnostic and paste the results here so I can confirm this is the issue. Thanks

whaleygeek, 2017-07-06:
If this is the case, there is not much I can do to fix it. It might be possible to catch this exception in bitio though and display a helpful hint suggesting what the cause of the problem is (and the usual way to recover from it). But I need to confirm if this is actually what is causing the problem first by you re-testing, and then decide the course of action from there.

whaleygeek, 2017-07-16:
@martinohanlon Have you managed to try this again to prove or deny my hypothesis?

martinohanlon, 2017-07-16:
Well, I wrote the 'ball game' on the Pi and didnt receive the error, so lets consider this 'resolved'.

closed 2017-07-16 by martinohanlon