whaleygeek/bitio #4: Raspberry Pi Python 2 error VTIMESerial not found

martinohanlon, 2017-06-20:
I received an error when trying to run bitio using Python 2 on the Pi, although I didnt receive the same error when running it on Python 2 on Windows:

```
Traceback (most recent call last):
  File "button.py", line 4, in <module>
    import microbit
  File "/home/pi/bitio/src/microbit/__init__.py", line 40, in <module>
    import serial
  File "/home/pi/bitio/src/microbit/serial/__init__.py", line 29, in <module>
    from serial.serialposix import Serial, PosixPollSerial, VTIMESerial  # noqa
ImportError: cannot import name VTIMESerial
```

whaleygeek, 2017-06-30:
Sounds like an unsatisfied package dependency in pyserial, presumably a bug in the version 2 distribution. I'll keep a lookout for it. When I do some Pi testing I'll see if I can do a quick fix to stop this happening, as I don't need any of the time services in what I do.

whaleygeek, 2017-07-06:
at the moment this is a wontfix, but I might upgrade the status later when I have more time to investigate properly.

DavidWhaleMEF, 2017-07-25:
I looked into this a bit today. I have improved the error messages in bitio to try to track this down.

DavidWhaleMEF, 2017-07-25:
I ran it on my Pixel OS (running in a VM on my Mac) and I don't see this error.

I note from the path of the exception that it has loaded the pyserial that is bundled with bitio rather than the system provided one.

However the inability to load VTIMESerial module is very odd.

DavidWhaleMEF, 2017-07-25:
@martinohanlon can you confirm what precise version of Python 2 you are running on your Raspberry Pi there please? I don't get this error when I run it up on my Pixel OS here.

whaleygeek, 2017-07-25:
@martinohanlon I had a look into this. It turns out that VTIMESerial is an 'adaptor layer' that the pyserial author put in to step over a linux ttys bug. It's a bit historic now, and this github issue suggests it is no longer used:

https://github.com/pyserial/pyserial/issues/82

So, I took out the reference to VTIMESerial (as my pycharm editor marked it as grey to say it was not in use anyway and a grep shows it was not used anywhere explicitly). Changed in this commit:

https://github.com/whaleygeek/bitio/commit/5e8713eab137aa586ebba45cf780263c70f171bc

I've asked @gemmamaylatham to try it on her PiTop OS next time she gets a chance and to report back here.

I've been unable to replicate this at this end on my Mac or on my Pixel OS. Struggling to find all the necessary bits required to get my Pi booting at the moment.

But I think this should fix it.

whaleygeek, 2017-07-25:
If it doesn't fix it, then you can temporarily turn on debug mode like this:

```
python counter.py debug
```

You will then get extra information at the point of failure. Added here: https://github.com/whaleygeek/bitio/commit/85287fef6678ee39ffed70eca53f3d31497f28f8

GemmaMayLatham, 2017-07-26:
Just successfully cloned and run the counter code in Python 3

GemmaMayLatham, 2017-07-26:
Python 2 working also

GemmaMayLatham, 2017-07-26:
This is in latest PiTop Os with updates just run

whaleygeek, 2017-07-26:
@gemmamaylatham great thanks for letting us know!

whaleygeek, 2017-07-26:
@martinohanlon This issue is now fixed - can you re-test at your end to make sure the problem has gone away for you too? Thanks!

whaleygeek, 2017-08-03:
@martinohanlon any update on whether this fixed your problem or not? Ta!

closed 2017-08-03 by whaleygeek

martinohanlon, 2017-08-07:
@whaleygeek retested...  This issue has now been resolved. Sorry for the delay.