whaleygeek/bitio #7: Change device to microbit

martinohanlon, 2017-07-05:
A minor issue.

When scanning you get messages like:

"No device has been previously been detected"
"remove Device then press enter"

Would it be 'nicer' to change "device" to "microbit" - It makes it more explicit.

whaleygeek, 2017-07-06:
Yeah, this is because the 'findport' utility that is embedded needs to be generic, as it is a package I drag and drop into lots of projects.

I'll have a think though if I can parameterise it from the containing package by passing in an override for the word it uses in the scan message, so it defaults to 'device' but can be overriden by the bitio/microbit package when it initiates the port scan, as a parameter override.

That way I can still use the findport package in other non-micro:bit projects unmodified (anyio in particular), but also make it in this case say micro:bit. So we'll get the best of both worlds.

whaleygeek, 2017-07-16:
Fixed by https://github.com/whaleygeek/bitio/commit/c0836f29904765112014ed831fdedd8f605bb071

whaleygeek, 2017-07-16:
FYI @simonmonk this makes a tiny change to the instructions that come up when scanning for a micro:bit - instead of saying 'device' it now says 'micro:bit'

Here is the sample trace of a scan for a micro:bit

```
connecting...
warning:Could not open the serial port that was remembered from last time
Scanning for serial ports
remove micro:bit, then press ENTER
scanning...
found 132 micro:bit(s)
plug in micro:bit, then press ENTER
scanning...
found 133 micro:bit(s)
found 1 new micro:bit
selected:/dev/tty.usbmodem1422
Do you want this micro:bit to be remembered? (Y/N)y
Your micro:bit has been detected
Now running your program
```

whaleygeek, 2017-07-16:
P.S. @martinohanlon @simonmonk this is fun too :-)

[image]

closed 2017-07-16 by whaleygeek