whaleygeek/bitio #11: Use with multiple simultaneous serial ports

whaleygeek, 2017-07-26:
@gemmamaylatham asked if it is possible to use bitio with a micro:bit while another serial port is connected to another device. e.g. to run concurrently with this project: https://github.com/whaleygeek/punchcard_reader

whaleygeek, 2017-07-26:
The portscan used in both these projects is similar, and was only really designed to detect one device. That means if you use two devices, the portscan.cache file will conflict.

Also, the punchcard reader project uses an older version of the portscan module that doesn't yet run in Python 3.

whaleygeek, 2017-07-26:
The punchcard reader code is not yet Python 3 safe: https://github.com/whaleygeek/punchcard_reader/issues/2

whaleygeek, 2017-07-26:
To make the both work together in the same Python program at the same time, I'll need to make the name of the portscan.cache file configurable, so that both packages could have a different name.

The new version of the portscan module in bitio has a configurable device name (micro:bit for example). We could probably use the device name as part of the cache file name (e.g. portscan.microbit and portscan.cardreader) to generate unique names so that the same device discovery method could be used for both in the same Python program.

It's not a huge change, but not something I want to rush.

titimoby, 2017-09-26:
using another device is cool, but using multiple micro:bit will be even more cool.
do you think it's doable while you code the initial goal of this issue fix?

whaleygeek, 2017-09-26:
It's definitely possible. But I need to think about the best way to surface this functionality, so that it is obvious to the user, and also so that it works as closely as possible to how the micro:bit APIs work (which is one of my goals for this package).

Give me a little bit of time to think about how I want to surface this, but I will report back here.

titimoby, 2017-09-26:
Take the time you need, it was just to note multi micro:bit will be as useful as multiple serial devices.