whaleygeek/bitio #17: radio support

whaleygeek, 2017-08-28:
The micro:bit has a radio module. I'd like to be able to, on the host machine, do this

import radio
radio.on()
radio.send('hello world')

I'd also like to be able to change the radio group and change the radio channel.

I'd also like to be able to set up a tight loop that notifies via a function when an incoming radio message arrives. I'd like a way to do this whereby the poll time of bitio is not too critical, so that messages are not lost.

I'd like to be able to run multiple instances of bitio connected to different micro:bits each tuned to different channels or listening on different groups, so that I can build multi-channel projects.

whaleygeek, 2017-08-28:
Similar 'module' issues need resolving as per https://github.com/whaleygeek/bitio/issues/16

whaleygeek, 2017-08-28:
Dependent on work being done by: https://github.com/whaleygeek/bitio/issues/11 for the multi-micro:bit work

robmarkcole, 2017-08-28:
Is it a 433MHz radio? Lots of projects on e.g hackster make use of this and would be nice to setup the microbit as a general purpose receiver/transmitter

whaleygeek, 2017-08-28:
The radio on the micro:bit uses the onboard Bluetooth hardware. It's the Nordic Gazel protocol, so it's at 2.4GHz - it's the super simple same radio interface you get in both python.microbit.org and makecode.microbit.org

whaleygeek, 2020-09-05:
Fixed by https://github.com/whaleygeek/bitio/commit/a6eae579057a943f3613aea551620cc4a69f48cf

closed 2020-09-05 by whaleygeek