whaleygeek/bitio #20: can't set serial port

titimoby, 2017-10-26:
(Ubuntu 17.04, user added too dialout group)
After flashing bitio.hex I launched counter.py but it loops through a cycle:

> No micro:bit has previously been detected
> Scanning for serial ports
> remove device, then press ENTER
> scanning...
> found 98 device(s)
> plug in device, then press ENTER
> scanning...
> found 99 device(s)
> found 1 new device
> selected:/dev/ttyACM0
> Do you want this device to be remembered? (Y/N)Y
> connecting...
> warning:Could not open the serial port that was remembered from last time
> Scanning for serial ports
> remove micro:bit, then press ENTER
> scanning...
> found 98 micro:bit(s)
> plug in micro:bit, then press ENTER
> scanning...
> found 99 micro:bit(s)
> found 1 new micro:bit
> selected:/dev/ttyACM0
> Do you want this micro:bit to be remembered? (Y/N)Y
> warning:Could not open the serial port that was remembered from last time
> Scanning for serial ports
> remove micro:bit, then press ENTER
> scanning...
> found 98 micro:bit(s)
> plug in micro:bit, then press ENTER
> scanning...
> found 99 micro:bit(s)
> found 1 new micro:bit
> selected:/dev/ttyACM0
> Do you want this micro:bit to be remembered? (Y/N)Y
> warning:Could not open the serial port that was remembered from last time
> Scanning for serial ports
> remove micro:bit, then press ENTER

whaleygeek, 2017-10-26:
Can you see if you can open the serial port with a terminal program logged in as the same user? e.g. miniterm, using /dev/ttyACM0

Also can you paste the contents here of your portscan.cache file that is in the same directory as counter.py

Finally, can you run it in debug mode like this:

```
python3 counter.py debug
```

and paste the output here

Thanks

whaleygeek, 2017-10-26:
P.S. debug is triggered from this line, so just put the word 'debug' on the command line and it will output more verbose debug info to the console.

https://github.com/whaleygeek/bitio/blob/master/src/microbit/__init__.py#L28

```
# Allow user to set debug flag on command line
if 'debug' in sys.argv:
    DEBUG = True
```

titimoby, 2017-10-26:
I could connect using "screen" but I didn't see anything outgoing (but I imagine I need to push commands before)

`thierry-laptop% python3 counter.py debug
Using path:/home/thierry/Dropbox/MixTeen/SoftShake2017/code/additional/microbit
Yay, I loaded the BITIO packaged pyserial
will reuse cache or scan for new port
No micro:bit has previously been detected
Scanning for serial ports
remove device, then press ENTER
scanning...
found 98 device(s)
plug in device, then press ENTER
scanning...
found 99 device(s)
found 1 new device
selected:/dev/ttyACM0
Do you want this device to be remembered? (Y/N)Y
connecting...
getting active serial port connection to micro:bit
warning:Could not open the serial port that was remembered from last time
Scanning for serial ports
remove micro:bit, then press ENTER`

titimoby, 2017-10-26:
I have no portscan.cache file, nowhere...

titimoby, 2017-10-26:
I wanted to understand what was wrong and I tried to launch counter.py from PyCharm and put a breakpoint in getSerial in __init__
... and it worked...

guess I need to learn pdb or other command line debugger to see what's different when I launch from my shell

titimoby, 2017-10-26:
Woooooow, that is weird.
I successfully used a breakpoint inside __init__.py and the program went smooth then.
The problem is only when I run it in a normal way...

whaleygeek, 2017-10-26:
Perhaps you don't have create access to the folder where it is trying to store portscan.cache?

That file would normally be created in the same folder as the script (counter.py), although that does assume that the current working directory is there too.

I usually open a terminal and cd to the folder where counter.py is, and run it from there, to be sure that cwd is where you expect it to be.

What was the output from adding debug to the command line, in the failing case?
Can you paste that here?

zeth, 2018-09-16:
I managed to get it to work on Ubuntu 18.04 by adding a print statement telling me what the exceptions were.

First I had error 13 (PermissionError), which was fixed with:

    sudo adduser zethg dialout

Then I had 16 (device or resource busy). This was because the built in dial up modem support was claiming the device. This was fixed with:

    sudo apt-get purge modemmanager

(Note, don't do this if you need to use a real dial up modem.)

Hope that helps anyone that gets to this issue on Google as I did.

xx

DavidWhaleMEF, 2018-09-16:
Super, thanks @zeth this is great detective work!

thebot75, 2019-07-11:
Thanks Zeth, had the same issue and this sorted it right out.  Cheers!