whaleygeek/bitio #23: MicroPython update required for motion sensor update

jaustin, 2018-10-02:
Now that MicroPython v1.0 is released (https://github.com/bbcmicrobit/micropython/releases/tag/v1.0.0) is it possible to update the bitio.hex file distributed in this repo to include the latest MicroPython build please?

DavidWhaleMEF, 2018-12-09:
It's a quick fix, but I then have to re-test all my book examples, to make sure there isn't any knock-on effects due to upgrading the version of MicroPython used. I plan to do this over the Christmas break.

whaleygeek, 2018-12-10:
As discussed with foundation: One thing to watch is some people have problems downloading files from github if they are not used to it. I might put a link on the bitio readme that goes straight to the latest bitio.hex and say 'right click this link, then save_as" to make it super easy!

whaleygeek, 2019-04-30:
I've been busy, and I don't want to release this until I have time to re-test all dependent code in my book. I have added a NEWSFLASH to the Readme advising users they can self-generate an updated bitio.hex in two easy steps using the online python editor.

whaleygeek, 2019-04-30:
User reported that this self-help fix works

https://twitter.com/thecloud_1/status/1123283256496066561

whaleygeek, 2020-12-12:
@jaustin I am doing some work over the winter break on bitio, to bring it up to date with some of the new features. On the 'newarch' branch I have an updated bitio.hex that was re-rolled from the latest python.microbit.org editor today, and this supports V1.5 and V2 features. I have tested it on a V2 and it works fine.

My plans are to pull over changes to the main branch when I have tested that all the examples work properly (as I don't want to break anything). I have also signposted the main branch readme to explain that the latest hex file that supports all devices can be retrieved from the newarch branch and included an easy link for people to follow.

There are other issues logged to capture the V2 work that I am doing.

https://github.com/whaleygeek/bitio/issues/31
https://github.com/whaleygeek/bitio/issues/32