whaleygeek/bitio #33: bitio.hex fails on latest firmware for micro:bit V2

mytechnotalent, 2020-12-12:
@whaleygeek as discussed when I flash bitio.hex on the latest firmware for V2 the display shows a frown face and then 5 2 1. I am unable to get a REPL or get any additional functionality.

whaleygeek, 2020-12-12:
The bitio.hex currently cached in this project is for a V1 micro:bit only.

For instructions on how to get an updated hex file that should work fine on all devices, just follow https://github.com/whaleygeek/bitio/issues/25#issuecomment-575646765 as bitio.hex is *just* a standard MicroPython build with an image display on start (to make the user experience better so you know the code is running, as by default a micro:bit does not display anything when it starts and this confuses users, thinking it hasn't worked).

I will update the cached bitio.hex in my latest V2 upgrade work, but this workaround should be fine for now (it's what I will do anyway, but I need to re-test all examples before pushing to the repo, and I probably will do that over the winter break)

mytechnotalent, 2020-12-12:
Thanks David!

Kevin Thomas
Senior Software Engineer in Test

For instructions on how to get an updated hex file that should work fine on all devices, just follow #25 (comment)<https://github.com/whaleygeek/bitio/issues/25#issuecomment-575646765> as bitio.hex is just a standard MicroPython build with an image display on start (to make the user experience better so you know the code is running, as by default a micro:bit does not display anything when it starts and this confuses users, thinking it hasn't worked).

mytechnotalent, 2020-12-12:
@whaleygeek I have confirmed it is working so I will close the issue.
```
from microbit import *

copyright = "bitio (c) 2017 David Whale"

def bitio():
    LOGO = Image("90000:90090:90909:00909:00090")
    display.show(LOGO)

bitio()
print(copyright)
```

closed 2020-12-12 by mytechnotalent