whaleygeek/bitio #34: Reports of radio.send() not working

whaleygeek, 2020-12-12:
There has been a report of the radio.send() not working. Need to investigate further.

mytechnotalent, 2020-12-12:
@whaleygeek see below code 

from receiving micro:bit V2:
```
from microbit import *
import radio
import time

while True:
    radio.on()
    
    incoming = radio.receive()
    print(incoming)
    time.sleep(1)
```

from transmitting bitio connect micro:bit V2:
```
import microbit

microbit.radio.config(group=132, queue=10)
microbit.radio.on()

microbit.radio.send('test')
```

The recv micro:bit V2 only gets `None` and does not get the `test` as expected. 

I have also tried commenting out the `microbit.radio.config(group=132, queue=10` as well.

whaleygeek, 2020-12-12:
The radio.receive() will repeatedly return None until a payload appears - this is a polled interface that is non blocking. It will do this repeatedly at 1ms intervals, which will probably flood the REPL with data, and your receiving terminal connected to the REPL will likely overflow and loose data.

You will also need the config line in the transmitter removed, so that both micro:bits are on the same default channel and default group (or add the same radio.config to the receiver so both ends parameters match).

Also, turn the radio on once and don't keep turning it on, as it is possible that a call to radio.on() reconfigures the hardware and packets may be loss during reconfiguration - you are currently reconfiguring the radio every 1ms.

I would recommend you use this code pattern in the receiver:

```
from microbit import *
import radio
import time

radio.on()

while True:
    incoming = radio.receive()
    if incoming:
      print(incoming)
```

mytechnotalent, 2020-12-12:
@whaleygeek I have made the changes you requested and I am still not receiving any data as I changed the receiving code as you suggested as well as removing the config line from the transmitter.

whaleygeek, 2020-12-12:
Ok. I'll fire up my micro:bit later and try this use-case out for you, and will report back.

whaleygeek, 2020-12-12:
Can you load some radio send/receive code directly onto those two micro:bits (without bitio) (e.g. first micro:bit sends a message when button A pressed) and micro:bit 2 displays an icon, waits 250ms then clears the screen when message received) to verify that you don't have a faulty radio on one of your devices?

mytechnotalent, 2020-12-12:
@whaleygeek confirmed the radios are both functioning normal.

MB 1:
```
from microbit import *
import radio
import time

radio.on()

print('Radio ON')

while True:
    display.show(Image.HAPPY)
    time.sleep_ms(250)
    incoming = radio.receive()
    if incoming:
      print(incoming)
      display.scroll(incoming)
```

whaleygeek, 2020-12-12:
MB2 stdout?

whaleygeek, 2020-12-12:
I am working on this branch: https://github.com/whaleygeek/bitio/tree/newarch

bitio.hex updated on that branch, going to do some radio tests now.

mytechnotalent, 2020-12-12:
Standard output.

whaleygeek, 2020-12-12:
I have just pushed my test code to the newarch branch - this works on a pair of V1 micro:bits fine (load the new radio.py on the host end, and the new for_microbit/radio_rx.py via python.microbit.org to the bitio device.

I will test this on a pair of V2's now and see what happens. But it seems the basic send functionality of bitio is working, I just need to see if there is a reason why it stops working on a V2 device.

For stdout, I meant your trace is MB2 stdout (not MB1 stdout as your comment stated?)

whaleygeek, 2020-12-12:
Here is my 'radio_rx' hex that I used to test with. I used bitio.hex in the newarch branch, for the transmitter.
[receiver.hex.zip](https://github.com/whaleygeek/bitio/files/5683666/receiver.hex.zip)

whaleygeek, 2020-12-12:
Yes, so I used the bitio.hex from the newarch branch, and the receiver.hex attached to this issue, loaded onto two V2 microbits, and it works.

I see this on the host console, updating once per second...

```
SEND
SEND
SEND
```

On the receiving micro:bit, I see a diamond flashing and then changing to a dot.

Same behaviour on a pair of V1 micro:bits with same setup.

A V1 micro:bit and a V2 micro:bit both in receiver mode, receive ok from a V2 microbit as bitio host sender.

A V1 micro:bit and a V2 micro:bit both in receive mode, receive ok from a V1 microbit as bitio host sender.

Hopefully these updated hex files and exact programs here, will give you a reproducible test case that you can replicate at your end.

mytechnotalent, 2020-12-12:
@whaleygeek I was able to reproduce.  I followed the exact setup and am experiencing the same.  Thank you David!  Will you be merging that branch with main?

whaleygeek, 2020-12-12:
@mytechnotalent Super, glad it works for you!

Yes, I still want to test that all the example programs still work, then I will merge the changes to main so that it is the default. But I will keep the newarch branch open, as I have other V2 features I want to add over the winter break.

Thanks for testing the code for me, it's really helpful to have others helping out!

closed 2020-12-12 by whaleygeek

mytechnotalent, 2020-12-13:
My pleasure good friend!