whaleygeek/bitio #35: Add unittest test suites

whaleygeek, 2020-12-13:
There is some possible issue with importing, that breaks unittest discover mode, that I need to investigate.

I will probably write some unittest suites for the newarch work, and that will flush out this problem.

This might also be related: https://github.com/whaleygeek/bitio/issues/22

mytechnotalent, 2020-12-13:
Thanks @whaleygeek I will keep an eye out for updates.

whaleygeek, 2020-12-13:
I have a single unit test working

https://github.com/whaleygeek/bitio/commit/40eff7cba7d43420b374942155181ddb44cd1c3a

Note that if I do this
```python3 -m unittest discover```

It runs my test but there are two errors, and a similar dump to yours.

if I do this instead, it works fine

```python3 tests.py```

NOTE:The added extract_ut_args() allows you to run selective tests

```python3 tests.py TestButton.test_ButtonA```

mytechnotalent, 2020-12-14:
@whaleygeek I am able to get my unittests to pass however when running the `python3 -m unittest discover` those two of yours are failing.  It is not a blocker by any means however would be great if we could get them working.