whaleygeek/bitio #43: Timeout error

happy-jin1234, 2023-08-10:
i'm using microbit 1.5v
and here is my code
```
import microbit
import pyautogui 
import pydirectinput
import time

while True:
    if microbit.button_a.is_pressed():
        pydirectinput.keyDown('s')
        time.sleep(0.2)
        pydirectinput.keyUp('s')
    elif microbit.button_b.is_pressed():
        pydirectinput.keyDown('w')
        time.sleep(0.2)
        pydirectinput.keyUp('w')
```
it runs correctly but few minutes after it's got an error
```
connecting...
Your micro:bit has been detected
Now running your program
Traceback (most recent call last):
  File "c:\Users\진\Downloads\bitio-master\src\counter.py", line 13, in <module>
    elif microbit.button_b.is_pressed():
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\진\Downloads\bitio-master\src\microbit\api.py", line 53, in is_pressed
    r = self.parent.cmd("print(%s.is_pressed())" % self.name)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\진\Downloads\bitio-master\src\microbit\api.py", line 30, in cmd
    r = self.repl.wait_response()
        ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\진\Downloads\bitio-master\src\microbit\repl\repl.py", line 168, in wait_response
    self.receive("OK", timeout=1, min_length=2)
  File "c:\Users\진\Downloads\bitio-master\src\microbit\repl\repl.py", line 82, in receive
    raise REPLException("Timeout trying to receive [%s]" % buffer)
microbit.repl.repl.REPLException: Timeout trying to receive [?bitio (c) 2017 David Whale
MicroPython v1.7-9-gbe020eb on 2016-04-18; micro:bit with nRF51822
Type "help()" for more information.
>>> ]
```

whaleygeek, 2023-08-10:
Looks like you pressed the reset button or the micro:bit rebooted, as the response includes the bootup message.

Check the reset button is not accidentally pressed, or perhaps you shorted 0V and 3V somehow and it caused a reboot.

happy-jin1234, 2023-08-15:
It is correct that the reset button was pressed. thank you