whaleygeek/bitio #44: REPL: could not get a prompt

RadoTheProgrammer, 2024-03-09:
I, i got an error when i try running `import microbit`

```
>>> import microbit
connecting...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/alain/RR/Rpy-mc/bitio/microbit/__init__.py", line 130, in <module>
    repl.to_raw()
  File "/Users/alain/RR/Rpy-mc/bitio/microbit/repl/repl.py", line 105, in to_raw
    raise REPLException("could not get a prompt")
microbit.repl.repl.REPLException: could not get a prompt
```

How to fix that error ?

whaleygeek, 2024-03-10:
The usual cause for this is where you have either not loaded the provided .hex file onto the micro:bit, there is a problem with the version of the .hex file, or you are connecting to a different COM port that doesn't have a micro:bit on it.

1.&2. You can regenerate the bitio.hex file that is cached in this repository (so that it has all the right hardware drivers for newer micro:bits) by drag-dropping it into python.microbit.org and then pressing the DOWNLOAD button - then flash that new .hex file to your micro:bit by drag-dropping it onto the MICROBIT drive that appears when you plug the micro:bit into the computer.

3. The COM port detected is stored inside a file called portscan.cache file, then follow the documented workflow in the README whereby you unplug and plug in the micro:bit and it discovers the port.

If you have issues with your USB lead, try this: https://support.microbit.org/support/solutions/articles/19000037633-what-type-of-usb-lead-do-i-need-for-the-micro-bit-

You can verify that your micro:bit is flashed/connected correctly by attaching it to your computer, going to python.microbit.org, connecting, and then opening the serial window, you should get a REPL prompt printed and be able to send commands like ```print("hello")``` and it will respond. If it does not respond in this way, bitio will not be able to communicate with the micro:bit

https://support.microbit.org/support/solutions/articles/19000135210-python-editor-guide#serial

whaleygeek, 2024-03-18:
This has been reported by another user too.

I suspect if you wind back the Python version on your host computer it might be a workaround. I suspect Python3.12 has broken the embedded Pyserial by deprecating some methods or something. We embeeded pyserial so that it was a zero-install package (file copy) on Raspberry Pi (for kids in schools) to make the install dependency free. It might be time to upgrade the embedded pyserial, or use a different way of deploying with dependencies I suspect.

I will try to investigate this further over the Easter hols